Harlem-Launch-Alliance/Stimulant #57: Stop adding to large integers

JonathanSafer, 2022-08-28:
Adding to large integers can be dangerous, as it risks unexpected behavior during an overflow. In `main.ino`, there is a line `delay = millis() + 3000;`. This could be problematic since `millis()` might be a very large number (on the verge of overflow). If that were the case, the statement a couple lines later, `delay > millis()`, might never occur.

To prevent this we should replace the former line with `delay = millis();`, and the latter statement with `delay - 3000 > millis()`

JohnS0819, 2022-08-29:
I'm not sure this is really an issue. There really shouldn't be any overflows (in realistic situations) here. `millis()` takes almost 50 days to overflow and on Teensy 4.1 both longs and ints are 32 bit, so overflowing by adding just 3000 seems unlikely.

`delay - 3000 > millis()` seems more problematic to me as it risks very easily underflowing.

JonathanSafer, 2022-08-29:
My mistake. That is an underflow risk. The proper statement should be `millis() - delay > 3000`. While you are correct that this overflow issue will almost certainly never show up, it is good to have proper procedure with time in place since we are using `micros()` at other points in this code.